kantlivelong/OctoPrint-PSUControl #191: Raspberry Pi 3 - PSUControl stopped working

MichiTheSailor, 2021-04-18:
Ciao

I'm Running OctoPrint and using PSUControl since log time and it worked for me really fine.

Two days ago I received an PSUControl Update from 0.1.11 to 1.0.0 and since this upgrade the PSUControll seems not be able any more to switch my Printer On and Off. 

I'm running Octoprint on an Raspberry Pi 3 Band using as PowerSwitching Output Pin 40 (GPIO 21).

I tried several solutions which I found in the issue report of others but unfortunatelly nothing worked for me.
Yesterday I also upgraded to PSUControl 1.0.1 withou any changes...

My current settings:
[image]
Alternativ:
[image]

[image]
[image]

Systeminfo:
[image]

Systemlog:
[octoprint.log-5.txt](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6330834/octoprint.log-5.txt)

The UI show, that PSU should be switched (using Sensing Internal)
[image]
[image]

But looking on the output PIN it still remains on 0V.

To fix I also tried o use addionally the external Plugin PSUControl RPi.GPIO, but also with this no success.

#### Printer model & used firmware incl. version
Due tr printer is not started not relevant, but I'm using CR10S Creality.

MichiTheSailor, 2021-04-18:
My Wiring digramm which works since years...
[image]

LouDnl, 2021-04-18:
I can confirm that this is the case. I have a Pi3b aswell and sensing stopped working since the new version. I already have the patch installed. Tried the sub plugin but that didn't work either.

kantlivelong, 2021-04-18:
@MichiTheSailor did you make any changes after upgrading?

kantlivelong, 2021-04-18:
If you haven't already be sure to check any GPIO related settings and ensure pin numbering is using BCM and not BOARD.

moarkaboom, 2021-04-18:
Exact same setup for me as described by MichiTheSailor (except I am using GPIO 3) with exact same result: Before the update, everything was working flawlessly.
Can you link to which logs you require to help you narrow down the error?

moarkaboom, 2021-04-18:
> If you haven't already be sure to check any GPIO related settings and ensure pin numbering is using BCM and not BOARD.

So this actually solved it for me. Prior to update, I was using Pin **3** (GPIO 2) in the settings, and it worked flawlessly. After the update, I had to change it to the correct GPIO, which is GPIO 2 for Pin 3. So now, with the correct BCM Numbering, it works again!

kantlivelong, 2021-04-18:
It appears that there is some pin conversion issue affecting some users.

MichiTheSailor, 2021-04-18:
Currently I have the same setting as before in use with 0.1.11 which worked.
As I wrote I tried several changes as suggested to solve like using System Commands instead of GPIO...
For System Commands I tried:
gpio -g write 21 1     (assumpion that the GPIO-Nr. should be used)
gpio -g write 40 1     (assumpion that the Pin-Nr. should be used)
Unlikely nothing worked ...
Afterwards I tried to use the Plugin, therefore installed RPi.GPIO and tried to config similar - No success.
Therefore deinstalled the plugin and and set back to GPIO Pin 40 as before (and shown above)
I also tried to use sensing Port-Pin 40 to grap the state, but this does not work and resulted in Errors in the Log, therefore I switched back to "internal" sensing

jachdrbm, 2021-04-18:
> So this actually solved it for me. Prior to update, I was using Pin **3** (GPIO 2) in the settings, and it worked flawlessly. After the update, I had to change it to the correct GPIO, which is GPIO [...]

I also did this and now the plug in works again (Raspberry Pi 3B+).

kantlivelong, 2021-04-18:
> I also tried to use sensing Port-Pin 40 to grap the state,

You need to use BCM(GPIO) numbering not BOARD.

abrenoch, 2021-04-18:
Just a note: like other users mentioned, this was resolved by manually switching the the BCM numbering (16 -> 23 for me). However, I did have to restart octoprint to get it actually working.

MichiTheSailor, 2021-04-18:
@kantlivelong Considering your feedback I changed (again because i tried it several times) to BCM 21.
[image]
[image]
Funny, now it worked - Very strange (and surprising :o) )   👍 

If I do the same using system Commands it did not work...  👎 
[image]

Strange...
But now the solution works - Thanks for your support

kantlivelong, 2021-04-18:
Would need logs for system command but not really worth doing IMO.

LouDnl, 2021-04-18:
In the end I reverted back to pre v1.0.0 as I uses one gpio for switching and sensing.

MichiTheSailor, 2021-04-18:
@kantlivelong  Attached a log where you can see not working system command.

I worked after our fix with GPIO 21 which worked fine - Thanks

I switched for you at 23:13 to system Commands.
Afterwards the Button in the Header changed in pressing, but the Pin GPIO 21 did not switch to high (5V).
I performed afterwards an Restart to see if this will change the result - But no, remains not working for System Commands.

Hope that help you
[octoprint.log](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6332223/octoprint.log)

If I can do further analysis for you let me know - Thanks from Vienna

MichiTheSailor, 2021-04-18:
Updated to 1.0.2 --> no change on the behaviour
Tried to use sensing gpio 21 as well: In this case I was not able any more to switch off. Therefore still using "Internal" because this is working best for me...

kantlivelong, 2021-04-18:
> but the Pin GPIO 21 did not switch to high (5V).

The Raspberry Pi uses 3.3V logic. Are you using a 5V relay?

> switched for you at 23:13 to system Commands.

Do they work in SSH? I just tried on GPIO12 and `gpio -g write 12 1` does not work for me. Wiring Pi was deprecated almost 2 years ago. http://wiringpi.com/wiringpi-deprecated/

MichiTheSailor, 2021-04-23:
> The Raspberry Pi uses 3.3V logic. Are you using a 5V relay?
You are right - See above, I use a transistor to switch the 5V relay...

kantlivelong, 2021-04-23:
> 
> 
> > The Raspberry Pi uses 3.3V logic. Are you using a 5V relay?
> > You are right - See above, I use a transistor to switch the 5V relay...

ah. right

So have you tried the latest PSUControl release with GPIO instead of SysCmd?

MichiTheSailor, 2021-04-23:
Yes, I installed PSUControl 1.0.3  
Using GPIO (BCM 21) it works fine
Using SystemCommands (gpio -g write 21 1) I can't switch the port

kantlivelong, 2021-04-23:
1.0.4 was released a few days ago but it won't fix your WiringPi problem

>Using SystemCommands (gpio -g write 21 1) I can't switch the port
WiringPi is deprecated and has been unmaintained for almost 2 years so I wouldn't expect it to work. You can try something like `echo on > /home/pi/test.txt` to see if SysCmd is working on your end.

By the sounds of it everything is working as it is supposed to Will keep this closed then but feel free to respond.

closed 2021-04-23 by kantlivelong

MichiTheSailor, 2021-04-23:
> By the sounds of it everything is working as it is supposed to. Will keep this closed then but feel free to respond.

Fully agree, because GPIO is working and System Commands should only be a workaround - Close Issue

Thanks for your investigation!