kantlivelong/OctoPrint-PSUControl #192: Sensing no longer working after update (and patch)

LouDnl, 2021-04-18:
#### What were you doing?
I updated the plugin and sensing no longer works.
Installed the patch and it still doesn't work.
Tried the sub plugin and still doesn't work.

1. Install new update
2. Install patch
3. Sensing doesnt work.

#### What did you expect to happen?
The printer to turn on and sensing to work and the power button turning green

#### What happened instead?
The printer turned on but sensing did not sense the printer and the button did not turn on.

#### Version of OctoPrint-PSUControl
1.0.1

#### Operating System running OctoPrint
OctoPi on Pi 3b

#### Printer model & used firmware incl. version
Ender 3 v2, self compiled Marlin v2 firmware.
This is not the issue, everything worked fine before you updated the plugin.

*MY OCTOPRINT.LOG IS SPAMMED WITH THE FOLLOWING ERRORS:*
2021-04-18 11:54:00,190 - octoprint.plugins.psucontrol - ERROR - 'sense'
2021-04-18 11:54:00,320 - tornado.access - WARNING - 409 GET /api/printer (::ffff:127.0.0.1) 15.56ms
2021-04-18 11:54:02,328 - tornado.access - WARNING - 409 GET /api/printer (::ffff:127.0.0.1) 10.44ms
2021-04-18 11:54:04,317 - tornado.access - WARNING - 409 GET /api/printer (::ffff:127.0.0.1) 11.17ms

kantlivelong, 2021-04-18:
Those errors are unrelated. Need full debug logs.

If you haven't already be sure to check any GPIO related settings and ensure pin numbering is using BCM and not BOARD.

LouDnl, 2021-04-18:
> 
> 
> Those errors are unrelated. Need full debug logs.
> 
> If you haven't already be sure to check any GPIO related settings and ensure pin numbering is using BCM and not BOARD.

Everything worked just fine before the update.
I use 1 gpio for switching and sensing. 
Eventually changed sensing to use system command with: gpio read 12
This changed by itself to Internal and now it works.

Its just not as quick in sensing as it was before.

kantlivelong, 2021-04-18:
Dupe of #182

closed 2021-04-18 by kantlivelong