kantlivelong/OctoPrint-PSUControl #193: Can't get GPIO sensing to sense

wick3r, 2021-04-18:
#### What were you doing?

1. Press on switch on PSU
2. _Raspberry Pi comes on, Printer comes on_
3. _After about 20 seconds, printer goes off_ _(side topic: If I connect printer to relay's "normally closed" rather than "normally open", it's the opposite: printer off, 20 seconds later printer on)_
4. _Octoprint has booted up. Lightning bolt is gray._
4. Pressed lightning bolt
5. _Printer comes on, but lightning bolt stays gray._
6. Pressed lightning bolt
7. _No result. Printer stays on, and lightning bolt stays gray_ (when I have sensing set to internal, printer goes off as it should)

#### What did you expect to happen?

When I press the lightning bolt and the printer comes on, I expect GPIO sensing to detect that 3.3v is coming in to GPIO pin 16, and make the lightning bolt green. I've tested with the multimeter that when the printer is on, 3.3v is coming out of the 3.3v pin from the Wifi port of the LCD, and when the printer is off, 0.0v comes out.

Screenshot of my PSU Control Plugin settings attached.

#### What happened instead?

The lightning bolt stays gray.

It looks like the log keeps saying "octoprint.plugins.psucontrol - DEBUG - isPSUOn: 0" which would make me think psucontrol knows there is 3.3v in GPIO 16. But it says that even if nothing is connected to GPIO 16.

#### Version of OctoPrint-PSUControl

1.0.2

#### Operating System running OctoPrint

Octopi

#### Printer model & used firmware incl. version

Ender 3 Pro
Marlin 2.0.7

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

https://pastebin.com/eipHy0KC

#### Wiring diagram

[image]

[Ender PSU Wiring.pdf](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6331304/Ender.PSU.Wiring.pdf)

wick3r, 2021-04-18:
wiring diagram attached
[Ender PSU [](Wiring.pdf](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6331356/Ender.PSU.Wiring.pdf))

wick3r, 2021-04-18:
wiring diagram attached
[Ender PSU Wiring.pdf](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6331358/Ender.PSU.Wiring.pdf)

kantlivelong, 2021-04-18:
`2021-04-18 09:04:22,874 - octoprint.plugins.psucontrol - ERROR - [Errno None] Line bias configuration not supported by kernel version 4.19.`

Update your kernel and should hopefully work.

wick3r, 2021-04-18:
Thanks. Noob here -- is kernel like my firmware?

kantlivelong, 2021-04-18:
I'm guessing you haven't installed system updates since install.

Make a backup first. Then follow https://www.raspberrypi.org/documentation/raspbian/updating.md

wick3r, 2021-04-18:
You guessed correctly. Installed the system update and the sensing works like a charm! Thank you.
Learned a lesson about system updates haha.

Now to prevent the 20-second power on-then-off of the printer during octopi startup, I changed the relay to "normally closed". In PSU Control I have to invert switching to get switching to work when the relay is "normally closed." Now the printer stays off during and also after octopi startup. Is there a way to make the printer come on automatically after octopi startup?

Markeno76, 2021-04-18:
I'll second the make a Backup First recommendation.  Trying to sort my issues, I attempted a full-upgrade on my Octopi 17.0.  It ended in errors updating about a Read Only Filesystem.  That left the installation non functional all commands unrecognized.  I am currently taking the control box apart that the Pi is stored in to reimage the SD card.  I do have a backup and all so it won't take long to get back up and running once I get the card out.

Does the option "Automatically turn PSU ON" work for that for you?

I will mention that I prefer the default state of the printer to be off when Octoprint is offline.  That way if Octoprint goes offline, the printer automatically turns off.  If for some reason the Pi goes down or looses connection the intention is for the printer to go off and not be left in a powered on heaters enabled condition.  That isn't to say that the Pi can't get in a locked state that it might leave the printer powered.

kantlivelong, 2021-04-18:
> Is there a way to make the printer come on automatically after octopi startup?

Not currently.

kantlivelong, 2021-04-18:
> Does the option "Automatically turn PSU ON" work for that for you?

AutoOn should work as that logic wasn't changed. It's triggered by defined GCODE. Printer must be connected.

kantlivelong, 2021-04-18:
I've added the kernel issue to [Wiki -> Troubleshootin -> Known Issues](https://github.com/kantlivelong/OctoPrint-PSUControl/wiki/Troubleshooting#known-issues)

Going to close this now.

closed 2021-04-18 by kantlivelong

wick3r, 2021-04-18:
Thanks for your help!