kantlivelong/OctoPrint-PSUControl #195: Update to 1.0.2 breaks connection to klipper-based printer

cmock, 2021-04-18:
#### What were you doing?

1. Update PSUcontrol to 1.0.2, Octoprint to 1.6.0rc2 (from 1.6.0rc1)

#### What did you expect to happen?

Printer should autoconnect on startup, and connect when I press the "Connect" button.

#### What happened instead?

Printer does not connect.

#### Version of OctoPrint-PSUControl

1.0.2

#### Operating System running OctoPrint

OctoPi 0.17.0

#### Printer model & used firmware incl. version

Kossel Mini running Klipper (from git).

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

https://pastebin.com/4gT6ziPv

#### Wiring diagram

Power is switched by the firmware via M81.

cmock, 2021-04-18:
Changing the "Switching method" setting away from "G-Code command", saving, and then switching back seems to have solved this. I noted this changes "_config_version:" from 3 to 4 in octoprint's config.yaml, and nothing else.

kantlivelong, 2021-04-18:
Interesting, it's claiming a setting didn't exist. Could you provide full logs though with the upgrade portion included?

kantlivelong, 2021-04-18:
Steps for capturing useful logs can be found at:
https://github.com/kantlivelong/OctoPrint-PSUControl/wiki/Troubleshooting#how-to-provide-logs

cmock, 2021-04-19:
Here's the full octoprint.log including the upgrade:
https://nextcloud.tahina.priv.at/index.php/s/rc5HqFa6WDm7FPs

There's a settings migration error at about 2021-04-18 16:33:03,307.

kantlivelong, 2021-04-19:
Thanks. Will look at this today.

kantlivelong, 2021-04-19:
Might be a dupe of #198

Can you try the build linked there?

cmock, 2021-04-19:
OK, so I did:
* stop octoprint
* change _config_version back to 3 under psucontrol in config.yaml
* verified the problem occured
* installed build from #198 

Now it works, _config_version was set to "4" automatically, printer connects.

new octoprint.log:
https://nextcloud.tahina.priv.at/index.php/s/SkdWiY4YSeiCPc9

kantlivelong, 2021-04-19:
@cmock thanks for reproducing the conditions on the test.

Dupe of #198

closed 2021-04-19 by kantlivelong