kantlivelong/OctoPrint-PSUControl #198: Sensing does not work at all on 1.0.2

evyd13, 2021-04-19:
#### What were you doing?
I have the plugin configured to send these commands on power on/off:
On:
`gpio mode 15 out && gpio write 15 0 && sleep 5 && echo "firmware_restart" > /tmp/printer &`

Off:
`gpio mode 15 out && gpio write 15 1`

That is with the internal sensing method. Used to work perfectly, but after the latest update this broke completely. I've tried the new GPIO plugin as well, doesn't work. Tried GPIO switching, doesn't work. No sensing method works either. I got it to turn my printer on, but it doesn't detect that it's on with any of the sensing methods.

1. ...
2. ...
3. ...

#### What did you expect to happen?
I expect the plugin to work just like it used to. I expect the button in the top to work.
#### What happened instead?
It didn't work.
#### Version of OctoPrint-PSUControl
1.0.2

#### Operating System running OctoPrint
OctoPi 0.17.0

#### Printer model & used firmware incl. version
Modified CR-10s PRO running Klipper. Somewhat latest version.

#### Attach or link DEBUG log files.
https://pastebin.com/kNsHtt11

#### Wiring diagram
I have a PSU with built-in switch (meanwell rsp-500-24) connected to BCM pin 14 on my Pi.

kantlivelong, 2021-04-19:
Thanks for providing everything. Going to take a look at this today.

blindley125, 2021-04-19:
Im am also having the same issue

kantlivelong, 2021-04-19:
Can you try with this build? https://github.com/kantlivelong/OctoPrint-PSUControl/archive/82194b4b229301801cd287dbe81b383b023658ab.zip

In Plugin Manager you can paste the link or upload the zip.

kantlivelong, 2021-04-19:
@evyd13

GPIO in PSUControl isn't working for you due to an the older kernel version. For PSUControl-RPi.GPIO you had the same pin value for both input and output. That worked on older builds but was actually a bug. Never even realized people were using the same pin number as it kind of defeats the purpose of a sense line. Seeing if there is a safe way to bring that back in though.
```
2021-04-19 12:22:22,424 - octoprint.plugins.psucontrol_rpigpio - DEBUG - GPIOMode: BCM
2021-04-19 12:22:22,424 - octoprint.plugins.psucontrol_rpigpio - DEBUG - onoffGPIOPin: 14
2021-04-19 12:22:22,425 - octoprint.plugins.psucontrol_rpigpio - DEBUG - invertonoffGPIOPin: False
2021-04-19 12:22:22,425 - octoprint.plugins.psucontrol_rpigpio - DEBUG - senseGPIOPin: 14
2021-04-19 12:22:22,426 - octoprint.plugins.psucontrol_rpigpio - DEBUG - invertsenseGPIOPin: False
2021-04-19 12:22:22,426 - octoprint.plugins.psucontrol_rpigpio - DEBUG - senseGPIOPinPUD: PULL_DOWN
2021-04-19 12:22:22,427 - octoprint.plugins.psucontrol_rpigpio - DEBUG - Cleaning up pin 14
2021-04-19 12:22:22,427 - octoprint.plugins.psucontrol_rpigpio - INFO - Running RPi.GPIO version 0.7.0
2021-04-19 12:22:22,427 - octoprint.plugins.psucontrol_rpigpio - INFO - Configuring GPIO for pin 14
2021-04-19 12:22:22,427 - octoprint.plugins.psucontrol_rpigpio - INFO - Configuring GPIO for pin 14
```

kantlivelong, 2021-04-19:
Will likely push out yet another release soon.

evyd13, 2021-04-19:
Thank you for fixing this so quickly ^.^

kantlivelong, 2021-04-19:
1.0.3 released.

closed 2021-04-19 by kantlivelong