kantlivelong/OctoPrint-PSUControl #199: Suggestion - Improving settings page for Raspberry Pi users

leoheck, 2021-04-19:
Is it possible to highlight (maybe in red) this line here?
A new line before of it would be nice too.
[image]

Also, it can have a link to the BCM/GPIO mappings
For instance, this one 
https://raw.githubusercontent.com/Gadgetoid/Pinout.xyz/master/resources/raspberry-pi-pinout.png
From this page (https://pinout.xyz/)

I was using the wrong number (3) there before of the last upgrade. Now I had to change to (BCM 2). Wich is the right thing. And it is working.

I just have one question. It looks like it takes too much time to change the image state. Is this normal?

kantlivelong, 2021-04-19:
Makes sense for now though I'm really trying to keep the core plugin platform agnostic.

GPIO and Internal sensing shouls be quick. System and Plugin based sensing may vary depending on what is being done.

leoheck, 2021-04-19:
Here it is using internal sensing with polling every 5s. I think it is the default.
I see, what is the difference between these options (System/Plugin/GPIO/Internal)

Off-topic:
It would be nice to have some tooltips explaining extra things on important fields at least.

kantlivelong, 2021-04-19:
Try using 1 second polling.

Docs are at https://github.com/kantlivelong/OctoPrint-PSUControl/wiki/Settings.

kantlivelong, 2021-04-19:
Added in 8d0be3113cd89df27fa6fc923c4a9c265c5157b5

leoheck, 2021-04-19:
I will try the 1s, for sure but after a couple of hours because it is in the middle of printing. However, I am not sure it will improve something since it looks like it is taking much than 5s to change its status.

kantlivelong, 2021-04-19:
If it's an issue def open a ticket with all required info in the template.