kantlivelong/OctoPrint-PSUControl #200: Not powering on after v1.0.0 until settings changed

Uinden, 2021-04-20:
So I have my relay on pin 24
```
    Physical/Board pin 18
    GPIO/BCM pin 24
    Wiring Pi pin 5
```
After v1.0.0 update I click power icon and nothing happens. Tried it multiple times.
Then I changed settings to pin 23 then back to 24 and everything started to work.
So if someone else have same problem, just change settings to other unused pin, click power icon few times then change it back to your pin and it should work.

Uinden, 2021-04-20:
Checked octoprint.log and here what I see related to `psucontrol`
```
2021-04-20 11:14:47,795 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2021-04-20 11:14:50,784 - octoprint.util.pip - INFO - Version of pip is 19.0.1
2021-04-20 11:14:50,789 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2021-04-20 11:14:50,790 - octoprint.util.pip - INFO - ==> pip ok -> yes
2021-04-20 11:14:50,853 - octoprint.plugin.core - INFO - Initialized 19 plugin implementation(s)
2021-04-20 11:14:50,924 - octoprint.plugins.psucontrol - INFO - No GPIO pins to convert.
2021-04-20 11:14:50,925 - octoprint.plugins.psucontrol - INFO - Removing Setting: GPIOMode
2021-04-20 11:14:51,904 - octoprint.plugin.core - INFO - 21 plugin(s) registered with the system:
```
```
2021-04-20 11:15:10,795 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2021-04-20 11:15:10,797 - octoprint.plugins.bedlevelvisualizer - INFO - OctoPrint-BedLevelVisualizer loaded!
2021-04-20 11:15:10,799 - octoprint.plugins.psucontrol - INFO - Using GPIO for On/Off
2021-04-20 11:15:10,804 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 24
2021-04-20 11:15:10,805 - octoprint.plugins.psucontrol - ERROR - [Errno 2] Opening GPIO chip: No such file or directory
2021-04-20 11:15:13,735 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://192.168.1.5/'}
```
And looks like this is when I tried to turn it on:
```
2021-04-20 12:02:13,417 - octoprint.plugins.psucontrol - INFO - Using GPIO for On/Off
2021-04-20 12:02:13,419 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 24
2021-04-20 12:02:13,422 - octoprint.plugins.psucontrol - ERROR - Invalid path type, should be string.
2021-04-20 12:02:27,509 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-20 12:02:27,511 - octoprint.plugins.psucontrol - ERROR - 'switch'
2021-04-20 12:02:31,099 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-20 12:02:31,100 - octoprint.plugins.psucontrol - ERROR - 'switch'
2021-04-20 12:03:02,353 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-20 12:03:02,356 - octoprint.plugins.psucontrol - ERROR - 'switch'
2021-04-20 12:03:03,792 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-20 12:03:03,795 - octoprint.plugins.psucontrol - ERROR - 'switch'
2021-04-20 12:03:10,744 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-20 12:03:10,747 - octoprint.plugins.psucontrol - ERROR - 'switch'
2021-04-20 12:11:15,342 - octoprint.plugins.psucontrol - INFO - Using GPIO for On/Off
2021-04-20 12:11:15,346 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 23
2021-04-20 12:11:22,991 - octoprint.plugins.psucontrol - INFO - Switching PSU On
```
After that I see logs without errors. Probably didn't log when I changed settings.
```
2021-04-20 12:11:26,204 - octoprint.plugins.psucontrol - INFO - Switching PSU On
```
```
2021-04-20 12:11:31,372 - octoprint.plugins.psucontrol - INFO - Using GPIO for On/Off
2021-04-20 12:11:31,373 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 24
2021-04-20 12:11:38,358 - octoprint.plugins.psucontrol - INFO - Switching PSU Off
2021-04-20 12:11:39,260 - octoprint.plugins.psucontrol - INFO - Switching PSU On
```
```
2021-04-20 12:11:43,976 - octoprint.plugins.psucontrol - INFO - Switching PSU Off
```
No more errors, everything works.

apogay37, 2021-04-20:
>Probably didn't log when I changed settings.

Oh wait, it did
`Configuring GPIO for pin 23`
Then log is complete.

kantlivelong, 2021-04-20:
Mind providing the full logs as requested? Particularly for the upgrade.

Uinden, 2021-04-20:
No errors in `plugin_softwareupdate_console.log`
```
2021-04-20 11:13:59,269   /home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/kantlivelong/OctoPrint-PSUControl/archive/1.0.3.zip --no-cache-dir
2021-04-20 11:14:04,342 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
2021-04-20 11:14:04,345 > Collecting https://github.com/kantlivelong/OctoPrint-PSUControl/archive/1.0.3.zip
2021-04-20 11:14:05,357 > Downloading https://github.com/kantlivelong/OctoPrint-PSUControl/archive/1.0.3.zip (193kB)
2021-04-20 11:14:09,634 > Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PSUControl==1.0.3) (1.5.3)
2021-04-20 11:14:10,808 > Collecting python-periphery (from OctoPrint-PSUControl==1.0.3)
2021-04-20 11:14:12,284 > Downloading https://files.pythonhosted.org/packages/00/2c/39f674e6cefd3ff8f2825d37efbb4f763eb0504ccb4405d05a498026845e/python_periphery-2.3.0-py2.py3-none-any.whl
2021-04-20 11:14:16,651 > Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.09.23 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PSUControl==1.0.3) (2021.2.4)
2021-04-20 11:14:16,652 > Requirement already satisfied: OctoPrint-FileCheck>=2020.08.07 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PSUControl==1.0.3) (2021.2.23)
2021-04-20 11:14:16,654 > Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PSUControl==1.0.3) (1.1.0)
.......
2021-04-20 11:14:16,717 > Requirement already satisfied: six in /home/pi/oprint/lib/python2.7/site-packages (from websocket-client<1,>=0.57->OctoPrint->OctoPrint-PSUControl==1.0.3) (1.12.0)
2021-04-20 11:14:16,718 > Installing collected packages: python-periphery, OctoPrint-PSUControl
2021-04-20 11:14:16,719 > Found existing installation: OctoPrint-PSUControl 0.1.11
2021-04-20 11:14:16,720 > Uninstalling OctoPrint-PSUControl-0.1.11:
2021-04-20 11:14:18,094 > Successfully uninstalled OctoPrint-PSUControl-0.1.11
2021-04-20 11:14:18,095 > Running setup.py install for OctoPrint-PSUControl: started
2021-04-20 11:14:22,636 > Running setup.py install for OctoPrint-PSUControl: finished with status 'done'
2021-04-20 11:14:22,641 > Successfully installed OctoPrint-PSUControl-1.0.3 python-periphery-2.3.0
```
And posted everything about psucontrol from `octoprint.log`
Can't post full log, there is broken plugin writing sensetive information in logs that can be used to access my printer. 

Look, there is difference. 
How it was:
```
2021-04-20 12:02:13,417 - octoprint.plugins.psucontrol - INFO - Using GPIO for On/Off
2021-04-20 12:02:13,419 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 24
2021-04-20 12:02:13,422 - octoprint.plugins.psucontrol - ERROR - Invalid path type, should be string.
2021-04-20 12:02:27,509 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-20 12:02:27,511 - octoprint.plugins.psucontrol - ERROR - 'switch'
```
It became:
```
2021-04-20 12:11:31,372 - octoprint.plugins.psucontrol - INFO - Using GPIO for On/Off
2021-04-20 12:11:31,373 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 24
2021-04-20 12:11:38,358 - octoprint.plugins.psucontrol - INFO - Switching PSU Off
2021-04-20 12:11:39,260 - octoprint.plugins.psucontrol - INFO - Switching PSU On
```
So **I suspect** there is config value somewhere that was something like `pin: 24` but needs to be `pin: "24"`  
You know better, try to set config value to number instead of string and check if error reproducing.

kantlivelong, 2021-04-20:
Were you using BCM mode previously?

`2021-04-20 11:14:50,924 - octoprint.plugins.psucontrol - INFO - No GPIO pins to convert`

might explain why no dev was selected. Likely something off in the update logic..

`2021-04-20 12:02:13,422 - octoprint.plugins.psucontrol - ERROR - Invalid path type, should be string.` is referring tothe GPIODevice value which likely didn't get set.

Uinden, 2021-04-20:
>Were you using BCM mode previously?

Honestly I do not know. I configured psucontrol long time ago and I don't remember which settings there was set. And I do not have any backups and I will regret that at some point.

kantlivelong, 2021-04-20:
I'll see if I can replicate it.

Uinden, 2021-04-20:
Hey! Turns out I do have backup from 2020-08-06 and I didn't even know.
Here is part from `config.yaml`

```
    psucontrol:
        GPIOMode: BCM
        _config_version: 3
        enablePowerOffWarningDialog: false
        idleTimeout: '30'
        onoffGPIOPin: '24'
        powerOffWhenIdle: true
        switchingMethod: GPIO
```

Uinden, 2021-04-20:
Oh, I see now.
There is check https://github.com/kantlivelong/OctoPrint-PSUControl/blob/258f6df4943b1349eec15cc011c02afd70b9ba75/octoprint_psucontrol/__init__.py#L683
`cur_GPIOMode == 'BOARD'` but my mode is BCM so its like "Then nothing to convert" but while converting it also creating `GPIODevice` https://github.com/kantlivelong/OctoPrint-PSUControl/blob/258f6df4943b1349eec15cc011c02afd70b9ba75/octoprint_psucontrol/__init__.py#L721 
So it did not created this setting.
**So there is bug - even if mode is BCM you still need to create `GPIODevice` setting.**

kantlivelong, 2021-04-20:
Yep!

3e956bafe1b1fe084b98c5cb5bb3ba85a4d7805e should fix it but will have to test before pushing. Hopefully that's the last of the settings migration issues.

kantlivelong, 2021-04-20:
Tested and confirmed. Merged to devel via f3d787beb03fddabca3af1d5ebcdce1ecc95e5e7

Going to release this as 1.0.4 now since it affects people updating. Hoping this is the last rapid update. :crossed_fingers:

closed 2021-04-20 by kantlivelong

Electronlibre2012, 2023-04-16:
Hello, i have exact the same issue today and i am under 1.0.4 version plugin

i have try uninstall/reinstall, same problem,

what can i do please?