kantlivelong/OctoPrint-PSUControl #206: Upload via API: Printer turns on but print does not start automatically

oerkel47, 2021-04-29:
#### What were you doing?

1. Octoprint running already (not controlled by psu control)
2. Printer is OFF (controlled by psu control)
3. in psucontrol  "turning the printer on prior to printing when uploading file from API" is activated

4. send a file from slicer via API with activated start print command

#### What did you expect to happen?

File uploads & printer gets turned on via psucontrol.
File starts printing.

#### What happened instead?

File uploads & printer gets turned on via psucontrol.
File is selected but print does not start.

#### Version of OctoPrint-PSUControl
1.05

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

https://pastebin.com/KaXuvYi3
The log might not be super helpful since its more the lack of stuff happening that is interesting.

#### my take on this problem

To make it work as expected it is necessary to make sure the printer is connected and then retrigger a print command. This is because the "original" print command gets lost if the printer is not 100% ready when the api file upload comes in.

#### What I tried: 
https://github.com/kantlivelong/OctoPrint-PSUControl/pull/205

Works in my case, but might not work in every case.

kantlivelong, 2021-04-29:
I really hate to be repetitive but the logs you linked are not complete. Check out the instructions in the wiki under troubleshooting.

kantlivelong, 2021-04-29:
https://github.com/kantlivelong/OctoPrint-PSUControl/wiki/Troubleshooting#how-to-provide-logs

oerkel47, 2021-04-29:
Do you miss the serial.log? I created another logfile. Logging was set to debug already.

Serial.log 
https://pastebin.com/3w1kUKdR
Octoprint.log:
https://pastebin.com/bLNULLqY

Interesting part should start at around 19:56

kantlivelong, 2021-04-29:
Thanks. The previous log wasn't from startup. :). Will look in a bit.

kantlivelong, 2021-04-29:
Can you try without Arc Welder and if that doesn't work try with all other 3rd party plugins disabled?

oerkel47, 2021-04-29:
Disabling arc welder doesn't change it. Will try more tomorrow. Thx for looking into it.

oerkel47, 2021-04-30:
Just tried with a fresh install with only psucontrol as 3rd party plugin. Same behaviour. 
The behaviour is the same as if the printer was off. It uploads and selects the file. I am quite sure the reason is the printer not being connected when the api command comes in.

Badn1nja, 2021-04-30:
> I can work around by waiting for the usbserial to show in octoprint then connecting manually but maybe a customizable delay to the connection back to the printer could assist here?

Found the setting... Post On Delay becomes 'Connect Delay' when Connect to printer automatically is set. try that that may help

oerkel47, 2021-04-30:
I am using that option already and connecting works, just the print doesn't start. 

Does it start printing for you? Or only connect?

kantlivelong, 2021-04-30:
Will see if I can narrow it down. Might just have to add a connection wait.

Badn1nja, 2021-05-01:
I set the post delay to 10 and this was perfect for my ender 5 Pro. I can have thr printer now idle and power and print automatically via cura

oerkel47, 2021-05-01:
Oh Wow. It works with 10 seconds delay! With 8 seconds apparently the timing is off but good enough for connecting.

kantlivelong, 2021-05-01:
Possibly just an environmental difference. I had a few people test prior to releaae with delays of 5s and below. Will see if can improve the handling though.

Badn1nja, 2021-05-01:
Yes I am using an esp8266 running wled on the pi so if the connect is run
too early octopi fails as it cannot connect to that and deems nothing is
suitable. It needs to have the second usb avalible before working 💪

kantlivelong, 2021-05-04:
So I've made a test build to see if adding a simple wait during the upload helps. Since I don't have a printer with this type of configuration I'm unable to test. If someone wants to try it out the link is below it can be installed from the Plugin Manager via URL. 

https://github.com/kantlivelong/OctoPrint-PSUControl/archive/refs/heads/enhance_api_print.zip

When testing please set the "Post On Delay" value to 0. Attach debug logs even if it works.

Thanks 🙂

oerkel47, 2021-05-04:
Hi.
It doesn't do the trick unfortunately. Additionally the slicer will throw an error now:
HTTP 500: {"error":"The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application."}`

Delay too low: error (this is new)
Delay 10s (which worked before): works

Log with 0s delay:
https://pastebin.com/35r0Gf5x

kantlivelong, 2021-05-04:
`NameError: name 'c' is not defined` whoops. Just pushed a change. Reinstall with the same URL.

oerkel47, 2021-05-04:
Ah. This works! Nice.
https://pastebin.com/DDpPpy8b

kantlivelong, 2021-05-04:
> 
> 
> Ah. This works! Nice.
> https://pastebin.com/DDpPpy8b

Odd, I don't see anything in the log for it. Make sure power is off and the printer is disconnected. Having the logs is really important since I can't test it myself. Thanks 👍

oerkel47, 2021-05-04:
Sorry, you are right. Didn't reset the post on delay to zero.

..doesn't work. 
https://pastebin.com/wMWGT0Rk

kantlivelong, 2021-05-04:
`serial.serialutil.SerialException: [Errno 6] could not open port /dev/ttyACM0: [Errno 6] No such device or address: '/dev/ttyACM0'`

Alright now we've got confirmation that it attempted to connect and failed because of the lack of delay. Previously early in the ticket you had a delay of 8s and it was connecting but not in a ready state so it wouldn't print. So let's try a value like 5s for the delay and see what happens.

oerkel47, 2021-05-04:
5 seconds runs into timeout.
6 seconds works.

Both cases are in the log.

https://pastebin.com/FgvF5LaY

On another note: 
I noticed that the symbol lags when using this feature now. PSUcontrol has determined the PSU to be on but the symbol doesn't change to green before the timeout or succes. This is not really important, since normally you wouldn't monitor the GUI anyway while using this feature. But I guess you might want to know anyway.

kantlivelong, 2021-05-04:
So that kind of fits in with the original timings where 8s was not being long enough. With your 6s plus the plugin waiting an additional 3s sounds like its working.

The UI not updating is expected with how this is implemented as the API call is stuck waiting for the printer to connect. 

I'm not 100% committed to this implementation so will see if I can come up with something better.

oerkel47, 2021-05-04:
Afterall an entry in the FAQ/Wiki to adjust the delay timing until it works is probably a good enough solution. I just hit an edge case with the timing..

swills, 2021-08-21:
I am hitting this issue too. Even after setting the delay to 30 seconds, I still see the issue. Printer is turned on, connected, print job exists, but doesn't start.

swills, 2021-09-21:
I tried [the change in the enhance_api_print branch](https://github.com/kantlivelong/OctoPrint-PSUControl/compare/enhance_api_print) and it worked great. Please merge. :smiley:

jfbauer432, 2022-01-28:
I just tried that change as well.  Worked as expected.

BeppeTemp, 2022-07-15:
I created a new issues about this problem, someone have found some fix ?

Badn1nja, 2022-07-15:
> I created a new issues about this problem, someone have found some fix ?

Can you increase the delay? 

The info should be above

oerkel47, 2022-07-22:
@BeppeTemp you can try if my fork works for you as it works differently in that regard: https://github.com/oerkel47/OctoPrint-PSUControl

kantlivelong, 2022-07-22:
@oerkel47 I plan to get back on this project at some point in the future. Likely near the end of the year.

BeppeTemp, 2022-07-25:
> @BeppeTemp you can try if my fork works for you as it works differently in that regard: https://github.com/oerkel47/OctoPrint-PSUControl

Works perfectly thanks a lot, why it's not merged in this repo ?