kantlivelong/OctoPrint-PSUControl #210: Plugins: HomeAssistant and Shelly

edekeijzer, 2021-05-10:
Just to let you know, I really liked the plugin idea so I have created two plugins:
[OctoPrint-PSUControl-HomeAssistant](https://github.com/edekeijzer/OctoPrint-PSUControl-HomeAssistant): control a Home Assistant switch entity.
[OctoPrint-PSUControl-Shelly](https://github.com/edekeijzer/OctoPrint-PSUControl-Shelly): control a Shelly relay either directly or through Shelly Cloud.
Please note that while I did test them, these are just first versions and not yet released through the plugin repo. Any feedback is greatly appreciated on the plugin issue trackers.

nibblerrick, 2021-05-13:
I just installed the shelly plugin and configured it with local connection and a shelly plug-S.
I took a quick look at the octoprint-logs and there wasn't anything, so it seems to work just fine.
Thanks both of you for this plugins, makes it really convenient :-)

kantlivelong, 2021-05-22:
These were released.

closed 2021-05-22 by kantlivelong