kantlivelong/OctoPrint-PSUControl #212: PSU Control switches off printer during running print job

Schnabulation, 2021-05-18:
#### What were you doing?

I had this twice in the past: Printing happily from within OctoPrint. Suddently PSU Control switches off the printer.

A look into octoprint.log shows:
`2021-05-18 15:51:09,437 - octoprint.plugins.psucontrol - INFO - Idle timeout reached after 60 minute(s). Turning heaters off prior to shutting off PSU.`
`2021-05-18 15:51:09,437 - octoprint.plugins.psucontrol - INFO - Heaters below temperature.`
`2021-05-18 15:51:09,437 - octoprint.plugins.psucontrol - INFO - Switching PSU Off`

I have no idea how PSU Control thinks that the printer is a) idle and b) the heaters are below temperature: I was printing a model in PETG and the nozzle was at 240° and the bed at 80°C.

PSU Control is configured to should the PSU off after 60 mins of idle with the heaters below 50°C.

#### Version of OctoPrint-PSUControl

OctoPrint: 1.6.1

PSU Control: 1.0.6

#### Operating System running OctoPrint

Ubuntu Server 20.04.2 LTS running on vSphere virtual machine.

#### Printer model & used firmware incl. version

Ender 3 V2

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

https://gist.github.com/Schnabulation/57bfbbac4d97aa2691f9499950cb957f

kantlivelong, 2021-05-18:
Hello,

According to the log your print failed due to serial error before PSU Control shut power off.

```
021-05-18 14:50:32,960 - octoprint.plugins.action_command_notification - INFO - Got a notification: Printing Layer 131 of 329
      
      
        2021-05-18 14:50:44,724 - octoprint.plugins.action_command_notification - INFO - Got a notification: Printing Layer 132 of 329
      
      
        2021-05-18 14:50:57,216 - octoprint.plugins.action_command_notification - INFO - Got a notification: Printing Layer 133 of 329
      
      
        2021-05-18 14:51:09,162 - octoprint.plugins.action_command_notification - INFO - Got a notification: Printing Layer 134 of 329
      
      
        2021-05-18 14:51:09,585 - octoprint.util.comm - ERROR - Unexpected error while reading from serial port
      
      
        Traceback (most recent call last):
      
      
          File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 500, in read
      
      
            raise SerialException(
      
      
        serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
      
      
        

      
      
        During handling of the above exception, another exception occurred:
      
      
        

      
      
        Traceback (most recent call last):
      
      
          File "/home/steve/.local/lib/python3.8/site-packages/octoprint/util/comm.py", line 3860, in _readline
      
      
            ret = self._serial.readline()
      
      
          File "/home/steve/.local/lib/python3.8/site-packages/octoprint/util/comm.py", line 6488, in readline
      
      
            c = self.read(1)
      
      
          File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 509, in read
      
      
            raise SerialException('read failed: {}'.format(e))
      
      
        serial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
      
      
        2021-05-18 14:51:09,590 - octoprint.util.comm - ERROR - Please see https://faq.octoprint.org/serialerror for possible reasons of this.
      
      
        2021-05-18 14:51:09,593 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Offline after error"
      
      
        2021-05-18 14:51:09,596 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
      
      
        2021-05-18 14:51:10,091 - octoprint.plugins.DisplayLayerProgress - INFO - Printing stopped. Detailed progress stopped.
      
      
        2021-05-18 14:56:45,791 - octoprint.server.heartbeat - INFO - Server heartbeat &lt;3
      
      
        2021-05-18 15:11:45,791 - octoprint.server.heartbeat - INFO - Server heartbeat &lt;3
      
      
        2021-05-18 15:26:45,792 - octoprint.server.heartbeat - INFO - Server heartbeat &lt;3
      
      
        2021-05-18 15:41:45,792 - octoprint.server.heartbeat - INFO - Server heartbeat &lt;3
      
      
        2021-05-18 15:51:09,437 - octoprint.plugins.psucontrol - INFO - Idle timeout reached after 60 minute(s). Turning heaters off prior to shutting off PSU.
      
      
        2021-05-18 15:51:09,437 - octoprint.plugins.psucontrol - INFO - Heaters below temperature.
      
      
        2021-05-18 15:51:09,437 - octoprint.plugins.psucontrol - INFO - Switching PSU Off
  ```

Schnabulation, 2021-05-19:
Oh... I didn't notice that! I do have some problems with my USB extension cord that sometimes drops connection.

Thank you for your help - closing this.

closed 2021-05-19 by Schnabulation