kantlivelong/OctoPrint-PSUControl #214: PSU Control Sends Power On Command with "Post On GCode" when Home Assistant is not reachable

Sakrapher, 2021-05-28:
Hi there,

I observed how the printer suddenly executed the "Post On GCode" several times in a row while printing. shortly before that, due to an update, I restarted my home assistant, which I used to connect the plugin. only after my Home Assistant was available again did the printer continue printing, only then based on my "Post On GCode" in a completely different place.

If the connection is lost, the "isPSUOn" status changes from True to False.

Since commands are executed during printing, these (in my example G1) trigger the switching on of the printer, which also triggers the execution of the "Post On GCode".

I think the problem arises from the fact that if the Home Assistant cannot be reached, the isPSUOn variable is set to False.

In the appendix I send all kinds of logs, recordings and my active configuration.

[psucontrol.log](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6562187/psucontrol.log)
[GCode-Viewer.txt](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6562188/GCode-Viewer.txt)
[image]

kantlivelong, 2021-05-28:
This occurred because you have "Automatically turn PSU On" enabled which turns on based on the trigger commands below it. Since your sensing method was inaccessible it assumed it's off which would be expected and as a result continuously tried turning the supply on then sending post on commands.

Based on your config it does not make sense to have "Automatically turn PSU On" anyway. Turn that off and the issue shouldn't occur during print. 

I have #213 opened to rename the option since the plugin has gotten a bit more complex since it was first introduced.

closed 2021-05-28 by kantlivelong

edekeijzer, 2021-06-03:
@kantlivelong this issue has been on my mind lately. The state is now off (False) if it cannot be retrieved. Could it be an option to start with an undefined state (None) and only explicitly setting it to False if the sensing method actually returns that? That would open the possibility to act differently to an unknown state than to a PSU that's explicitly detected as being turned off.

kantlivelong, 2021-06-03:
> @kantlivelong this issue has been on my mind lately. The state is now off (False) if it cannot be retrieved. Could it be an option to start with an undefined state (None) and only explicitly setting [...]

I did start toying with the idea of an unknown state but it's been more of a cosmetic thing to inform the user thus far. Basically as you said it would just allow a `None` value to be interpreted as Unknown. It wouldn't really help with this scenario though because the option that was enabled (autoOn) involves turning the supply on if specific GCODE is sent. Even if the state was Unknown the supply should be turned on because the option essentially says "hey I need power to execute these commands properly". If unknown support was implemented then I think the right way to handle that would be to just fail the job entirely. Granted if the controller is being switched then autoOn shouldn't even be used as it's a chicken and egg scenario anyway. As #213 mentions I will change the wording at some point to make it more clear on what it does.