kantlivelong/OctoPrint-PSUControl #215: PSU Control turn off the PSU after changing any configuration and clicking save

gmlupatelli, 2021-06-02:
#### What were you doing?
1. With Octoprint up and running, click on the PSU control button on the title bar to turn on the printer.
2. After the control board is up and running, open the PSU Control configuration in OctoPrint Settings and change any parameter (e.g.: toggle "Turn off when an unrecoverable firmware or communication error occurs")
3. Click the Save button

#### What did you expect to happen?
I expected that the PSU Control configuring would be saved without affecting the Octoprint current work or restarting OctoPrint,

#### What happened instead?
After clicking save, the PSU control turns off the PSU. However, the status button is still flagging the PSU as on. 

#### Version of OctoPrint-PSUControl
OctoPrint 1.6.1 / Python 2.7.16

PSU Control (1.0.6)

#### Operating System running OctoPrint
OctoPi 0.17.0

#### Printer model & used firmware incl. version
Ender 3 with an SKR Mini E3 v1.3 running Marlin 2.0.7

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG
<script src="https://gist.github.com/gmlupatelli/44ed55261537d9d13b67ce9797a26085.js"></script>

kantlivelong, 2021-07-21:
Hi @gmlupatelli,

If anything I'd agree that config changes that aren't related to GPIO shouldn't cause a reset of the GPIO pin. This is actually how the behavior was in older versions. Since it's a low priority issue I'll keep the ticket up and tackle it at some point.

As for the out of sync status, that can be solved by using a sensing method other than Internal as it's just a basic bit flipper.

Rneil-ca, 2021-08-15:
This is a big pain for myself.  If I am doing a big print that needs a color change or could be stopped by the filament sensor, I like to disable the Automatically turn the PSU off when idle, as it will time out and shutdown, causing me to lose my reference.  I sometimes forget to do this before I start my print and now it I do it after it powers off the printer when saving.  You used to be able to do this and now you cannot.  

What would actually be nice is if turning the automatic turn  PSU off when idle could be changed in the API, without resetting the power of course.

Thanks for all your work on this project,
Ryan

0rphon, 2023-08-29:
same issue. i just caused a print to fail by accident because of this