kantlivelong/OctoPrint-PSUControl #219: Idle Power Off Problem after cold / hot start

Vitalic66, 2021-07-07:
Setup PIGS to control GPIO .

turn on system command: pigs w 21 1
turn off system command: pigs w 21 0

sensing via RPI.GPIO Plugin -> Pin 20 (hardware jumper connected to pin 21)

Commands and sensing work without problems.

If Pi just reboots everything runs well, but if pi is cold booted (completely powered off) or hot booted (restart from sleep state) the PSU starts and after several seconds (sometimes 8, sometimes 90) PSU turns printer off (GPIO 21 low).

In logfiles I can see that state is asked by PSU Plugin and after some seconds idle power off function is executed because idle time (10 minutes) is reached.

Tried several ways to turn PSU On (build in auto on function fails). 

Scripted in rc.local and also external script executed by crontab. Always idle power off time is reached after some seconds.

OctoPrint 1.6.1
OctoPi 0.18.0
PSUControl 1.0.6
PSUControl - RPI.GPIO 1.0.4

#### Printer model & used firmware incl. version

Creality CR10-V3 Marlin FW 2.0.8.1a

Vitalic66, 2021-07-07:
Found the problem from other logs, while I've seen this.

The PSU turns on, the wifi gets connection, then a time sync happens. The larger the off time was before, the faster the idle times is reached.

After boot PSU On
```
2021-07-07 21:17:55,263 - octoprint.plugins.psucontrol - DEBUG - Polling PSU state...
2021-07-07 21:17:55,263 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
```

Wifi connected and time synchronized, real time between both events few seconds.
```
2021-07-07 21:29:05,827 - octoprint.plugins.psucontrol - INFO - Idle timeout reached after 10 minute(s). Turning heaters off prior to shutting off PSU.
```

I'll try to run my script after wifi connected and time synced, this should work as workaround. Maybe you'll have time to edit the thread function where idle time is compared.

kantlivelong, 2021-07-07:
There isn't really anything for me to do here. [This](https://raspberrypi.stackexchange.com/questions/94635/how-can-i-delay-the-startup-of-systemd-services-until-the-datetime-is-set-no-rt#95195) might help you.

closed 2021-07-21 by kantlivelong

flipaz, 2021-12-31:
Would it be better to use the time elapsed since boot for the idle timer? I think it would solve this issue.