kantlivelong/OctoPrint-PSUControl #220: Post On Delay stalls the print when PSU device is offline

dominik-korsa, 2021-08-14:
#### What were you doing?

I have a SONOFF outlet switch with Tasmota flashed on it. I started my print (using CURA with the Octoprint plugin and PSU Control configured). At some point during the print the SONOFF device becomes unreachable (due to my Wi-Fi router, not something to do with OctoPrint or this plugin). 

#### What did you expect to happen?

Printing should continue despite the PSU being unreachable (not off).

#### What happened instead?

When the SONOFF becomes unreachable the print is veeeery slow, as OctoPrint tries to activate the PSU on every command and waits for Post On Delay.

#### Version of OctoPrint-PSUControl

OctoPrint-PSUControl 1.0.6

#### Operating System running OctoPrint

OctoPrint 1.6.1
Python 3.7.3
OctoPi 0.18.0

#### Printer model & used firmware incl. version

Longer LK4 Pro

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

I'm unable to provide the log at the moment, ping me if you really need it

### Possible fixes

- Don't activate the delay if the printer is already connected despite the PSU being unreachable. **Doesn't work if the printer is being back-powered from the Raspberry PI**
- Stop activating the delay after several failed connection attempts
- Don't activate the delay when PSU is turned on by a Trigger Command and PSU is unreachable

dominik-korsa, 2021-08-14:
Related: #214

kantlivelong, 2021-08-14:
https://github.com/edekeijzer/OctoPrint-PSUControl-HomeAssistant/issues/4#issuecomment-846396693

anjok, 2021-09-30:
It was very annoying to track the cause of the stalled prints down to this problem at all I did was not use the plug and instead just plug the printer in directly. 

At the very least spew out something in the console and not only in the logs. And I would opt for the third option above.