kantlivelong/OctoPrint-PSUControl #221: Issue with system Sensing System Command

Gleb1982, 2021-08-19:
#### What were you doing?
Orange Pi Zero 2 with Debian.
Am using system command as a switching method (gpio write 2 1) to turn_off and to turn_on (gpio write 2 0), all works as expected.
But use of system command for sensing ( gpio read 2) doesn't work. Did some testing, the plugging only understands 'true' or 'false'. Used such command to generate Boolean values (gpio read 2 | python -c "print (input()==0)" ), unfortunately Python returns True/False and not true/false, so it didn't help. 

1. Set  Sensing Method as system command.
2. Put (false) instead of the command see psu state.
3. Put (true) instead of the command see psu state.
4. Put (False) instead of the command see psu state.
5.3. Put (True) instead of the command see psu state.
can do the same with 0 or 1

#### What did you expect to happen?
Ideally the plugin should understand 0,1, false, true, False, True. 
#### What happened instead?

#### Version of OctoPrint-PSUControl
PSU Control (1.0.6)

#### Operating System running OctoPrint
 Debian Linux 4.9.170-sun50iw9

#### Printer model & used firmware incl. version
Climber 7

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG
none

electronicm, 2021-08-31:
i had the exact same problem. `gpio read` gave me 0 or 1. The plugin understood it, according to debug log, but as 127 (?).
I was able to fix it with `gpio read [PIN] | grep [expected return value]`.

Not the nicest way, but at least it works for now.

kantlivelong, 2021-09-01:
This option does not parse the _output_ but instead the _return code_. If using python you need to use sys.exit().

https://github.com/kantlivelong/OctoPrint-PSUControl/wiki/Settings#sensing

closed 2021-09-01 by kantlivelong