kantlivelong/OctoPrint-PSUControl #230: Pi Power Button and OctoPrint-PSUControl

arut16, 2021-12-04:
Hi all,

I'm using an RPi4 with an Octopi/Octoprint instance on it for 3D printer. In Octoprint software, i'm using 
OctoPrint-PSUControl plugin to control a 3v3 relay on GPIO21 and through it power on and down 3D printer with Raspberry. And without the physical power button, this working as it must.

I successfully installed the power-button on GPIO3 **[from this repo](https://github.com/Howchoo/pi-power-button)** and powering on and shutdown of raspberry is working well through the momentary button.

BUT. Then I send a command to close the relay and power on the printer through octoprint. The relay closes and the printer power on and at the same time raspberry seems to "receive" also the command to power off like if I push the new power button but I didn't touch it !

There must be a kind of interference of signal between [Pi Power Button](https://github.com/Howchoo/pi-power-button) and OctoPrint-PSUControl. But I don't know where to start in investigations...

Any help will be very kind !

#### Version of OctoPrint-PSUControl
1.0.6
#### Operating System running OctoPrint
[octoprint-systeminfo-20211204191905.zip](https://github.com/kantlivelong/OctoPrint-PSUControl/files/7654413/octoprint-systeminfo-20211204191905.zip)

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

[octoprint.log](https://github.com/kantlivelong/OctoPrint-PSUControl/files/7654418/octoprint.log)

kantlivelong, 2021-12-07:
This sounds more like a de-bounce or EMI issue on your physical switch. If this plugin is not configured to touch GPIO3(physical switch) then it wont touch it.

arut16, 2021-12-07:
> This sounds more like a de-bounce or EMI issue on your physical switch. If this plugin is not configured to touch GPIO3(physical switch) then it wont touch it.

Thanks for your reply. What do you suggest to do ?

kantlivelong, 2021-12-07:
Does the issue occur if you disconnect the switch from GPIO3?

arut16, 2021-12-07:
> Does the issue occur if you disconnect the switch from GPIO3?

If I disconnect, the button from GPIO3, when I can't not anymore power on/off Raspberry of course. Relay on GPIO21 working correctly to power on/off the printer.
The goal is to have a functional button on GPIO3 to power on/off the raspberry AND a working relay what is not interfere with GPIO3.

kantlivelong, 2021-12-07:
I'm going to have to direct you to the [OctoPrint Community Forums](https://community.octoprint.org) as this really isn't a PSUControl issue. If things are working correctly with the switch disconnected from GPIO3 then it is most likely a debounce or EMI issue as previously mentioned, not a PSUControl issue.