kantlivelong/OctoPrint-PSUControl #232: Doesn't work with active low modules

luizbgomide, 2021-12-20:
That is not a PSU Control issue per se, but could be a problem to many people.

#### What were you doing?

Trying to deactivate a low trigger 5V relay (VCC  to +5, GND to ground and, and signal to a GPIO pin).

#### What did you expect to happen?

That the relay would open when the GPIO was high.

#### What happened instead?

It doesn't open, because the 3.3V from gpio is not high enough, so the defender relay remains always active no matter the GPIO pin state.

I'm not sure if there is some way to use a low active relay with raspberry without some additional electronics (like what!?), but it is better to know that in advance...

kantlivelong, 2021-12-20:
Pretty common issue. I've seen some guides out there to mod 5V relays to work with 3.3V logic. Really it's up to the user to buy the appropriately spec'd relay.

luizbgomide, 2021-12-20:
Yes, it is very common. Which is why I think that maybe a notice warning of that possibility (or even a link to a "solution"), otherwise the user will be totally at lost unless he knows what is looking for to diagnose the issue... :/

kantlivelong, 2021-12-20:
https://github.com/kantlivelong/OctoPrint-PSUControl/wiki/Troubleshooting#common-issues

closed 2021-12-20 by kantlivelong