kantlivelong/OctoPrint-PSUControl #236: Power-off settings are ignored when print finishes, no delayed power-off

fnsign, 2022-04-20:
I run PSUControl with the TPLink subplugin (settings see attached screenshots, latest releases).
When a print finishes, the PSU is powered off imediatelly instead of waiting for the set time and temp.
Interestingly, when the printer is not printing and just idling, then the PSU is powered off correctly after the set time (the nozzle is of course cold and under the set temperature).

#### What were you doing?

1. Installed latest versions of PSUControl and PSUControl-TPLink 
2. Setup plugins (see screenshots)
3. Powered on printer via bolt icon in UI (means the plugin detects the TP-Link plug correctly). 
4. Started a print

#### What did you expect to happen?
After the print is finished the printer should stay powered on for 30 min until the nozzle temperature cooled down to 40C and then power off. This is for avoiding heat creep and clogging.

#### What happened instead?
Instead the PSU is powered off imediatelly as soon as the print finished. The settings for power off delay and temp are ignored.

#### Version of OctoPrint-PSUControl
PSUControl: 1.0.6
TPLink: 1.0.1

#### Operating System running OctoPrint

OctoPi: 0.17.0
Octoprint: 1.7.3

#### Printer model & used firmware incl. version

Creality Ender 3 with SKR E3 Mini v1.2 on latest Marlin 2.0.x bugfix

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

see attachments

[image]
[image]
[octoprint.log](https://github.com/kantlivelong/OctoPrint-PSUControl/files/8522535/octoprint.log)

kantlivelong, 2022-04-20:
Hello,

The logs appear to be incomplete. Follow https://github.com/kantlivelong/OctoPrint-PSUControl/wiki/Troubleshooting#how-to-provide-logs with serial logging enabled.

kantlivelong, 2022-04-20:
It's not entirely clear from the log but do you have an `M81` in either your gcode file or in your printcomplete gcode scripts in OctoPrint? 

```
2022-04-20 17:23:26,069 - octoprint.util.comm - INFO - Finished in 854.130 s.
2022-04-20 17:23:26,071 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Finishing"
2022-04-20 17:23:26,106 - octoprint.printer.standard.job - INFO - Print job done - origin: local, path: PLA_USB_dupont_Adapter_011_7m_0.2mm_205C_ENDER3BLTOUCH.aw.gcode, owner: fozi
2022-04-20 17:23:26,217 - octoprint.plugins.psucontrol - INFO - Switching PSU Off
2022-04-20 17:23:26,217 - octoprint.plugins.psucontrol - DEBUG - Switching PSU Off Using PLUGIN: psucontrol_tplink
```

The timing is a little off but may be due to comms being dropped.

`2022-04-20 17:23:36,621 - PSUControl: ok`

fnsign, 2022-04-20:
Shame on me...indeed there was a `M81` remainder in the "After print job completes" section in Octoprint. I looked everywhere else for such stuff, but not there. It's a remainder from the plugin I used before. Sorry for bothering you with my stupidity and thanks a lot for your prompt support! I'll close the issue, too.

closed 2022-04-20 by fnsign