kantlivelong/OctoPrint-PSUControl #242: Post On GCode Script - not executing

calvinbui, 2022-06-15:
#### What were you doing?

Turn on my lights with my 3D printer through the Post On GCode Script section. I am using the TP-Link Smartplug plugin to enable sending GCode commands to power on and power off the lights. 

The commands are either `M80/M81 192.168.3.23/1` or `@TPLINKON/@TPLINKOFF 192.168.3.23/1`. I can execute these commands from the OctoPrint terminal successfully.

The Gcode however is not executed during power on based on the octoprint and serial logs. The GCode for Pre Off however is executing successfully. 

I am powering on and off the printer via the nav bar lightning button.

[image]

#### What did you expect to happen?

The Gcode command to be executed.

They are executed when turning off the printer via the nav bar button

#### What happened instead?

The Gcode was not executed.

#### Version of OctoPrint-PSUControl

1.0.6

#### Operating System running OctoPrint

OctoPi RPi4 image

```
OctoPrint 1.8.1 Python 3.7.3 OctoPi 0.18.0
```

#### Printer model & used firmware incl. version

Prusa MK3S+

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

[octoprint (1).log](https://github.com/kantlivelong/OctoPrint-PSUControl/files/8910245/octoprint.1.log)
[serial (3).log](https://github.com/kantlivelong/OctoPrint-PSUControl/files/8910247/serial.3.log)

enishi1357, 2023-08-07:
Did you solve this?

Crono141, 2024-03-04:
I have this same problem, but with the pre-off commands.

calvinbui, 2025-01-22:
i've moved to https://plugins.octoprint.org/plugins/tplinksmartplug/

closed 2025-01-22 by calvinbui