kantlivelong/OctoPrint-PSUControl #244: Print starting on and receive gcode file, but not start print

BeppeTemp, 2022-07-14:
#### What were you doing?

**Initial situation:** Octoprint server tuned on (indipendente from PSU Control), Printer turned off (controlled from PSU Control, with HomeAssistant Plugin)

1. Press "Upload and Print" on Prusa Slicer
2. PSU Control start printer and wait 10 second (my post on delay setting)
3. Octoprint connect to printer and receive gcode file

#### What did you expect to happen?

After I press upload and print on Prusa Slicer, Octoprint turn on printer, wait "post delay time" (set to 10) receive the gcode file and start the print.

#### What happened instead?

After I press upload and print on Prusa Slicer, Octoprint turn on printer, wait "post delay time" (set to 10) receive the gcode file but don't start the print.

**Version of OctoPrint-PSUControl:** 1.0.6

#### Operating System running OctoPrint
I installed octoprint on a raspberyy Pi 4 2 Gb with Raspberry Pi Imager (Prebuild Octoprint Image)

#### Printer model & used firmware incl. version
Prusa MK3S+ firmware version 3.11.0

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

[serial.log](https://github.com/kantlivelong/OctoPrint-PSUControl/files/9113814/serial.log)
[octoprint.log](https://github.com/kantlivelong/OctoPrint-PSUControl/files/9113815/octoprint.log)

Badn1nja, 2022-07-15:
Hi Beppe,

It looks like Prusa might be slower to initiate, can you try a 60 or 120 second delay to see?

Is the printer on AMC0?

BeppeTemp, 2022-07-15:
What is AMC0 ?

I tried with 120 seconds but nothing, simply when I press "upload and print" the print turn off after 120 seconds, after that the gcode is uploaded but print not start

closed 2026-10-15 by BeppeTemp