kantlivelong/OctoPrint-PSUControl #252: Latest OctoPrint update broke the "Automatically turn PSU OFF when idle"

akamakamak, 2022-10-03:
#### What were you doing?

1. After upgrading to   OctoPrint 1.8.3/OctoPrint 1.8.4 the printer no longer shuts off when idle
2. I reverted back to OctoPrint 1.8.2 and everything works
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Version of OctoPrint-PSUControl

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

mdeeter, 2022-10-05:
I fixed mine (Octoprint 1.8.4) using my Ender 3 S1 by adding `M155` to the `ignore commands` option in the plug in.

[image]

akamakamak, 2022-10-06:
Thank you Michael.
I have tried it but it does not work for me.

courtpl, 2022-11-22:
I have exactly the same behaviour.
The printer no longer shuts off when idle

ClassicGOD, 2023-01-30:
If you are using LightsOut plugin remember to add the gcode commands used by this plugin to the ignored command list. LightsOut sends the set gcode command (to turn the lights off) every configured period of inactivity (not just once) if the PSU Control idle timeout is longer it will never see printer as idle even if the sent gcode is only "virtual" gcode for a plugin like GCODE System Commands. 

If it still does not trigger for you observe the Terminal to see what other gcode is sent to your printers.

courtpl, 2023-01-30:
I d'ont use this LightsOut plugin

anthonielamay, 2023-02-27:
Was there ever a resolution for this?  Recently flashed the new Octopi 1.0.0 and the "Automatically turn PSU OFF when idle" does not seem to work anymore.  Using a Prusa MK3S+.  Octopi connected via RPi4/USB.  Checked terminal; it does not seem to be sending anything to the printer.  Just the printer sending temp messages.

akamakamak, 2023-02-27:
> Was there ever a resolution for this? Recently flashed the new Octopi 1.0.0 and the "Automatically turn PSU OFF when idle" does not seem to work anymore. Using a Prusa MK3S+. Octopi connected via [...]

Can you downgrade the OctoPrint with the latest OctoPi?
For me the last working is OctoPrint 1.7.3.

kantlivelong, 2023-02-27:
https://raw.githubusercontent.com/kantlivelong/OctoPrint-PSUControl/master/.github/ISSUE_TEMPLATE.md

Also include serial logs.

akamakamak, 2023-02-27:
> https://raw.githubusercontent.com/kantlivelong/OctoPrint-PSUControl/master/.github/ISSUE_TEMPLATE.md
> 
> Also include serial logs.

Is the add-on working for you?

kantlivelong, 2023-02-27:
> > https://raw.githubusercontent.com/kantlivelong/OctoPrint-PSUControl/master/.github/ISSUE_TEMPLATE.md
> > Also include serial logs.
> 
> Is the add-on working for you?

My printing environment is offline at the moment.

courtpl, 2023-03-12:
Definitively,  I have exactly the same behaviour.
The printer no longer shuts off when idle

browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/110.0.0.0 Safari/537.36 Edg/110.0.1587.69
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: True
connectivity.enabled: True
connectivity.online: True
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: True
env.hardware.cores: 4
env.hardware.freq: 1400.0
env.hardware.ram: 780967936
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 3 Model B Plus Rev 1.3
env.plugins.pi_support.octopi_version: 0.17.0
env.plugins.pi_support.throttle_check_enabled: True
env.plugins.pi_support.throttle_check_functional: True
env.plugins.pi_support.throttle_state: 0x80000
env.python.pip: 20.2.3
env.python.version: 3.7.3
env.python.virtualenv: True
octoprint.last_safe_mode.date: unknown
octoprint.last_safe_mode.reason: unknown
octoprint.safe_mode: False
octoprint.version: 1.8.7
printer.firmware: Prusa-Firmware 3.12.1 based on Marlin
systeminfo.generated: 2023-03-12T20:40:00Z
systeminfo.generator: zipapi

kantlivelong, 2023-03-12:
> Definitively, I have exactly the same behaviour. The printer no longer shuts off when idle
>
> browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) [...]

https://raw.githubusercontent.com/kantlivelong/OctoPrint-PSUControl/master/.github/ISSUE_TEMPLATE.md

edwork, 2023-03-13:
I began having this issue after installing the plugin on a fresh installation of the RasPi image based install. I realized this plugin wasn't working and subscribed to this thread.  

Recently I realized that it started working again not having changed anything related to the plugin directly but I speculate it was after installing one or more additional plugins. 

I've listed all installed plugins below, I highly suspect Octolapse was the package that "fixed" the Auto Off functionality. 

```
Backup Scheduler
Enclosure Plugin
Filament Manager
Firmware Check
Firmware Updater
Heater Timeout
HomeAssistant Discovery
MQTT
MultiCam
OctoPod Notifications
Octolapse
PSU Control
PSU Control - Home Assistant
Preheat Button
PrintTimeGenius Plugin
Printer Dialogs
Printer Notifications
Slic3r
Telegram
Themeify
```

akamakamak, 2023-03-13:
> I began having this issue after installing the plugin on a fresh installation of the RasPi image based install. I realized this plugin wasn't working and subscribed to this thread.
>
> Recently I [...]

I had the Octolapse plugin install all along.

kumters, 2023-05-14:
I have PRUSA MINI and MK3S. There were no problems with PSU Control on MINI at all. Some months ago, I discovered the same issue as these guys on my MK3S. I have tried many things (based on this topic) without success. I tried to enable debug and check logs - nothing in the logs (it needs more detailed debug logs - by the way).

Today I decided to do some more debugging again, and based on this "Octolapse info", I checked my plugins and discovered that I had installed and enabled it on MINI but installed and disabled it on MK3S. **I enabled Octolapse on MK3S and Woala, and it started to work.** If @kantlivelong wants to do some deep debugging and testing, I'm ready to help:)

kumters, 2023-05-15:
OK, I was happy too early:( more printing and some testing = it works after a full octoprint reboot. It will stop working after the print.