kantlivelong/OctoPrint-PSUControl #253: Feature request.

jtannenba, 2022-11-26:
Not a problem, but a request.

Request a button/click or ?  for maintenance so printer won't shut down 
I know we can disable the pluggin, but a button on the title bar would be
convenient.

Thanks

jeloneal, 2022-12-29:
Great idea! I was just about creating a new issue for the same use case, but had a different idea how to achieve it:

I can only refer to my Prusa Mini, but when doing maintenance like changing filament or calibrating first layer, I normally disconnect the printer from octoprint via printers LCD menu. If I forget to turn off the PSU off function it will turn off the printer.

**My idea was if we could prevent turning the PSU off, when the printer is not connected to octoprint.** What do you think about that?

jtannenba, 2022-12-29:
I don't know, my printer is always connected to Octoprint. Or do you mean when the printer is offline?  If yes, would be great.

jeloneal, 2022-12-29:
Yes, by not connected I meant that Octoprint is has not connected to the printer. The cable is of course connected all the time though.

jtannenba, 2022-12-30:
Yep!  Would make maintenance easier.  I've had the PSU turn it off when I was working my Ender3 V2.