kantlivelong/OctoPrint-PSUControl #254: PSUControl doesn't work or is inconsistent in Docker installation

4rihant, 2022-12-31:
#### What were you doing?

1. Installed Octoprint via [prind](https://github.com/mkuf/prind), and turned up the containers
2. Installed PSUControl Plugin
3. Configured the PSUControl plugin according to my wiring. 

#### What did you expect to happen?
To be able to turn off and on the relay via PSU Control provided UI

#### What happened instead?

It doesn't turn on. It did turn on once and I was able to turn the PSU on and off, but then it randomly stopped working again and I haven't gotten it to work again. 

To make sure that my wiring is correct, I have tested my setup by using `raspi-gpio`  on the terminal via SSH to set the GPIO pin (16) to output and then driving it low to turn my relay on. 

#### Version of OctoPrint-PSUControl

1.0.6
#### Operating System running OctoPrint

Raspian 11 Bullseye

#### Printer model & used firmware incl. version

Ender-5, using Klipper

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

https://gist.github.com/4rihant/f7c3a2aa71f0a0782e1add3fef97162e

kantlivelong, 2023-03-06:
This is likely either a container issue or a wiring issue. Have you tested on bare metal?

jain-blueid, 2023-03-07:
Thanks for the response @kantlivelong, after many issues I ended up abandoning both docker's installation and Octoprint in favour of Klipper using KIUAH. This issue may be closed.

4rianton, 2023-03-07:
Hi @kantlivelong , thanks for your response. I just gave up on this after spending a lot of time. I moved to Klipper and Mainsail using KIUAH for now. This issue may be closed.

closed 2023-03-07 by 4rianton