kantlivelong/OctoPrint-PSUControl #259: Printing is extraordinarily slow/problematic if power supply unavailable

fix-digital, 2023-02-02:
#### What were you doing?

Fresh install of OctoPC; installed PSUControl + Tasmota sub-plugin as substitute for Tasmota plugin.  It is possible the issue is with the sub-plugin, I don't have any other equipment to test.

I then moved the printer and disconnected the physical Tasmota plug, but left the plugin(s) active.  I assumed the system would simply sent a power on/off command and otherwise mind its own business.

#### What did you expect to happen?

I expected prints to progress at a normal speed.

#### What happened instead?

Printing slowed almost to a halt.  3 hours to print a calibration cube, many print attempts failed completely with the head moving a fraction of a mm per minute.

I note that I traced the issue back to this plugin through [this Octoprint.org thread.](https://community.octoprint.org/t/octopi-sloooooow-printing-not-your-average-slow/15990) which also includes some logs I think.

#### Version of OctoPrint-PSUControl

PSU Control 1.0.6
PSU Control - Tasmota 1.0.1

#### Operating System running OctoPrint

ArchLinux/[OctoPC](http://nuclear-takeout.byethost7.com/Step1.html) not LiveCD)
OctoPrint 1.8.6

#### Printer model & used firmware incl. version

Creality CR10v3, Marlin 2.1.2 (I think)

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

I have a print going and not inclined to turn the problematic plugin back on, but my current log is literally 20MB of

2023-01-31 11:48:28,549 - octoprint.plugins.psucontrol - INFO - Auto-On - Turning PSU On (Triggered by G2)
2023-01-31 11:48:28,550 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2023-01-31 11:48:28,569 - octoprint.plugins.psucontrol_tasmota - ERROR - Unable to communicate with device. Check settings.
2023-01-31 11:48:28,711 - octoprint.plugins.psucontrol_tasmota - ERROR - Unable to communicate with device. Check settings.
2023-01-31 11:48:28,730 - octoprint.plugins.psucontrol - INFO - Auto-On - Turning PSU On (Triggered by G2)
2023-01-31 11:48:28,730 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2023-01-31 11:48:28,747 - octoprint.plugins.psucontrol_tasmota - ERROR - Unable to communicate with device. Check settings.
2023-01-31 11:48:28,870 - octoprint.plugins.psucontrol_tasmota - ERROR - Unable to communicate with device. Check settings

...interspersed with occasional real commands.

FrederikNS, 2023-02-11:
I'm experiencing the same issue, but with the Home-Assistant plugin. So it seems to be a general issue when there's a problem communicating with the upstream system.

To me it seems like every command gets delayed by 10 seconds or so. Printing something will have the nozzle move at normal speed to the end of the gcode instruction, then stop for 10 seconds, then move at normal speed for the next instruction, then wait for 10 seconds. So on and so forth...

FrederikNS, 2023-02-11:
Duplicate of #243

ulope, 2023-06-30:
Similar issues with other sub-plugins: #262, #243