kantlivelong/OctoPrint-PSUControl #262: Print stutters while Homeassistant is unavailable

MrJoki007, 2023-03-06:
Im using the PSU Control and PSU Control Homeassistant plugin, to turn my printer automaticaly off.
If my homeassistant gets unavailabe my printer starts to stutter.

1. Start print
2. Turn off homeassistant
3. ...

#### What did you expect to happen?
Print continues as normal
#### What happened instead?
Print stutters
#### Version of OctoPrint-PSUControl

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

ulope, 2023-06-30:
Same problem here. 
I think there are multiple issues here:

1. There's no need to try "turning on" the printer while it's clearly already printing (i.e. serial connection is established and state is printing)
2. The network request to HASS shouldn't block the whole octoprint main thread.

ulope, 2023-06-30:
Similar issues with other sub-plugins: #259, #243

AJolly, 2023-12-21:
I get this issue even when HA is available, and everything has a wired ethernet connection.  Was wondering wtf had happened to my printer.

pooh22, 2025-03-11:
The problem is that prints often stop or get paused for a long time, with sometimes a tiny movement.

I've been using 1.11.0rc versions and I've seen this happening mostly in rc3 and now in rc4.

During a pause I watched the logs on the RPI and it seems octoprint logs this in the syslog:

```
Mar 11 20:23:41 octopi octoprint[2534]: 2025-03-11 20:23:41,613 - octoprint.plugins.psucontrol - INFO - Switching PSU On#033[0m
Mar 11 20:23:56 octopi octoprint[2534]: 2025-03-11 20:23:56,788 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.#033[0m
Mar 11 20:23:56 octopi octoprint[2534]: 2025-03-11 20:23:56,837 - octoprint.plugins.psucontrol - INFO - Auto-On - Turning PSU On (Triggered by G1)#033[0m
Mar 11 20:23:56 octopi octoprint[2534]: 2025-03-11 20:23:56,838 - octoprint.plugins.psucontrol - INFO - Switching PSU On#033[0m
Mar 11 20:24:12 octopi octoprint[2534]: 2025-03-11 20:24:12,001 - octoprint.plugins.psucontrol - INFO - Auto-On - Turning PSU On (Triggered by G1)#033[0m
Mar 11 20:24:12 octopi octoprint[2534]: 2025-03-11 20:24:12,002 - octoprint.plugins.psucontrol - INFO - Switching PSU On#033[0m
Mar 11 20:24:27 octopi octoprint[2534]: 2025-03-11 20:24:27,189 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.#033[0m
Mar 11 20:24:27 octopi octoprint[2534]: 2025-03-11 20:24:27,250 - octoprint.plugins.psucontrol - INFO - Auto-On - Turning PSU On (Triggered by G1)#033[0m
Mar 11 20:24:27 octopi octoprint[2534]: 2025-03-11 20:24:27,251 - octoprint.plugins.psucontrol - INFO - Switching PSU On#033[0m
Mar 11 20:24:42 octopi octoprint[2534]: 2025-03-11 20:24:42,417 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.#033[0m
Mar 11 20:24:42 octopi octoprint[2534]: 2025-03-11 20:24:42,487 - octoprint.plugins.psucontrol - INFO - Auto-On - Turning PSU On (Triggered by G3)#033[0m
Mar 11 20:24:42 octopi octoprint[2534]: 2025-03-11 20:24:42,489 - octoprint.plugins.psucontrol - INFO - Switching PSU On#033[0m
Mar 11 20:24:57 octopi octoprint[2534]: 2025-03-11 20:24:57,652 - octoprint.plugins.psucontrol - INFO - Auto-On - Turning PSU On (Triggered by G1)#033[0m
Mar 11 20:24:57 octopi octoprint[2534]: 2025-03-11 20:24:57,653 - octoprint.plugins.psucontrol - INFO - Switching PSU On#033[0m
Mar 11 20:25:12 octopi octoprint[2534]: 2025-03-11 20:25:12,836 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.#033[0m
Mar 11 20:25:12 octopi octoprint[2534]: 2025-03-11 20:25:12,892 - octoprint.plugins.psucontrol - INFO - Auto-On - Turning PSU On (Triggered by G1)#033[0m
```

So it may be related to the PSU control plugin. I have this configured with the additional plugin with an interface to home assistant.

I will switch of the plugins for PSU control, to see if that makes any difference.