kantlivelong/OctoPrint-PSUControl #267: Issie with OctoPrint-TPLinkSmartplug using Mk4 and TPLink PLug

kevman28, 2023-06-08:
#### What were you doing?

1. Use Octopi with Prusa MK4 and TPLink plugin to control power to printer
2. Set Idle Timer
3. Idle timer expires but the power doesnt turn off automatically. 

#### What did you expect to happen?
wanted to tplink plug to be powered off after idle time
#### What happened instead?
plug stay on. 
#### Version of OctoPrint-PSUControl
Embedded in this plugin (https://github.com/jneilliii/OctoPrint-TPLinkSmartplug) v 1.03 of that plugin not sure if that helps. 
#### Operating System running OctoPrint

OctoPi 1.9

kantlivelong, 2023-06-08:
https://raw.githubusercontent.com/kantlivelong/OctoPrint-PSUControl/master/.github/ISSUE_TEMPLATE.md

kantlivelong, 2023-06-08:
https://github.com/kantlivelong/OctoPrint-PSUControl/wiki/Troubleshooting#how-to-provide-logs

Also if you're not actually using PSUControl/PSUControl-TPLink then no need and you can close this out.

closed 2023-06-08 by kantlivelong

BeppeTemp, 2023-12-05:
Same issue here, some news ?