kantlivelong/OctoPrint-PSUControl #271: Idle poweroff is not working on prusa MK4 with Firmware 5.1.2 (heatbreak X temp and target temp causes problems)

netinetwalker, 2024-01-22:
On the prusa mk4 idle poweroff is not working correct because of the temp target for cooling fan of the heatbreak is not 0 and should not changed from the target 36.00

Current g-code log of prusa mk4:
```
T:25.00/0.00 B:23.96/0.00 X:23.95/36.00 A:30.73/0.00 @:0 B@:0 HBR@:0
T:25.00/0.00 B:23.93/0.00 X:23.94/36.00 A:30.75/0.00 @:0 B@:0 HBR@:0
T:25.00/0.00 B:23.96/0.00 X:23.94/36.00 A:30.81/0.00 @:0 B@:0 HBR@:0
```

octoprint.log
```
2024-01-21 03:51:55,823 - octoprint.plugins.psucontrol - INFO - Idle timeout reached after 30 minute(s). Turning heaters off prior to shutting off PSU.
2024-01-21 03:51:55,823 - octoprint.plugins.psucontrol - DEBUG - Heater tool0 already off.
2024-01-21 03:51:55,824 - octoprint.plugins.psucontrol - DEBUG - Heater bed already off.
2024-01-21 03:51:55,824 - octoprint.plugins.psucontrol - INFO - Turning off heater: X
2024-01-21 03:51:59,140 - octoprint.plugins.psucontrol - DEBUG - Polling PSU state...
```

dirty quick fix: (I filter X in the /octoprint/util/comm.py) LINE 6414: 
```python
def parse_temperature_line(line, current):
    """
    Parses the provided temperature line.

    The result will be a dictionary mapping from the extruder or bed key to
    a tuple with current and target temperature. The result will be canonicalized
    with :func:`canonicalize_temperatures` before returning.

    Arguments:
        line (str): the temperature line to parse
        current (int): the current active extruder

    Returns:
        tuple: a 2-tuple with the maximum tool number and a dict mapping from
          key to (actual, target) tuples, with key either matching ``Tn`` for ``n >= 0`` or ``B``
    """

    result = {}
    max_tool_num = 0
    for match in re.finditer(regex_temp, line):
        values = match.groupdict()
        sensor = values["sensor"]
        if sensor == "X":
            # skip X sensor
            continue
        if sensor in result:
            # sensor already seen, let's not overwrite stuff
            continue
```

#### What were you doing?

Idle poweroff settings activated but printer remains powered.

#### What did you expect to happen?
Printer turns off

#### What happened instead?
Printer remains powered

#### Version of OctoPrint-PSUControl

PSU Control (1.0.6)
PSU Control - Shelly (1.0.2)

#### Operating System running OctoPrint

OctoPrint 1.9.3 - Python 3.9.2OctoPi* 1.0.0cam (build 2023.10.09.154319) on raspbian bullseye

#### Printer model & used firmware incl. version

Prusa MK4  
Prusa-Firmware-Buddy 5.1.2+13478 (Github) 

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

netinetwalker, 2024-01-22:
Looks like #267 is the same Issue and the solution from the dedicated OctoPrint-TPLinkSmartplug Plugin with a heater filter list can adapted here too -> https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues/336  

Sorry for the duplicate

netinetwalker, 2024-01-23:
QUICK FIX v2 small filter plugin: filter_heatbrake_temperatures_prusa_mk4.py
```python
# coding=utf-8

def filter_heatbrake_temperatures(comm, parsed_temps):
    parsed_temps.pop("X", None)
    return parsed_temps

__plugin_name__ = "Filter Prusa MK4 heatbrake Temperatures"
__plugin_version__ = "0.0.1"
__plugin_description__ = "A quick heatbrake temperatures filter plugin for Prusa MK4"
__plugin_pythoncompat__ = ">=2.7,<4"
__plugin_hooks__ = {
    "octoprint.comm.protocol.temperatures.received": filter_heatbrake_temperatures
}
```

jurassic73, 2024-04-30:
Has there been progress with this being implemented so other plugins can reference for control?

mogorman, 2024-05-02:
for my prusa xl its "X5" instead of "X"

jurassic73, 2024-05-07:
> QUICK FIX v2 small filter plugin: filter_heatbrake_temperatures_prusa_mk4.py
>
> ```python
> # coding=utf-8
>
> def filter_heatbrake_temperatures(comm, parsed_temps):
>     parsed_temps.pop("X", None) [...]

Do you create a file in ~/oprint/lib/python3.9/site-packages called filter_heatbrake_temperatures_prusa_mk4.py with this as contents then reload Octoprint?

wseifert, 2024-05-11:
I have also this issue, but using this fix does not solve it. The only result is that on the Octodash screen the nozzle temperature stays on the last one and does not decrease. 
My env is MK4 & MMU3 FW 6.0, Octoprint 1.10, Octodash 2.3.1
Any other hints to solve?

DavyGD, 2024-07-07:
I have the same issue, the plugin fixes it but another plugin uses the data that is being filtered so it's not very usable for me. Otherwise works perfectly. A filter for what the plugin ingests would be great.

oxivanisher, 2025-05-16:
I did a quick stab at this, but I had not the time to investigate on how to setup the dev environment for testing. And I am not sure that it would be merged anyways, seeing that there are several open PRs which are very old and the last development happend 4 years ago. Anyways, this might help somebody (it did not work for me, but I think its going in the right direction): https://github.com/oxivanisher/OctoPrint-PSUControl

akamakamak, 2025-12-31:
Nice work!
Works on my XL and Core ONE.

matonb, 2026-10-19:
Another data point: this isn't limited to Prusa firmware. Any non-standard sensor key in the temperature data triggers it, including garbage from serial line noise.

Setup: OctoPrint 1.11.8, PSU Control 1.0.6, Marlin printer on `/dev/ttyS0` (Pi GPIO UART). A corrupted serial line (OctoPrint warned about a null byte / data corruption) left a fake sensor `o` in the temperature data. When the idle timeout fired, the power-off sequence stopped at that sensor:

```
2026-10-18 18:12:53,033 - octoprint.plugins.psucontrol - INFO - Idle timeout reached after 10 minute(s). Turning heaters off prior to shutting off PSU.
2026-10-18 18:12:53,033 - octoprint.plugins.psucontrol - INFO - Turning off heater: o
...
ValueError: heater must match "tool", "tool([0-9])", "bed" or "chamber": o
```

`Switching PSU Off` was never logged, and the PSU stayed on until it was switched off by hand the next morning.

Suggested fix in `_wait_for_heaters()`: only act on valid heater names (`tool\d*`, `bed`, `chamber`), and/or wrap `self._printer.set_temperature()` in a try/except. That way a single bad entry can't stop the PSU from turning off.

jeffgregx2, 2026-10-22:
> 13\. Another data point: this isn't limited to Prusa firmware. Any non-standard sensor key in the temperature data triggers it, including garbage from serial line noise.
>      Setup: OctoPrint 1.11.8, PSU Control 1.0.6, Marlin printer on `/dev/ttyS0` (Pi GPIO UART). A corrupted serial line (OctoPrint warned about a null byte / data corruption) left a fake sensor `o` in the temperature data. When the idle timeout fired, the power-off sequence stopped at that sensor:
>      ```
>      2026-10-18 18:12:53,033 - octoprint.plugins.psucontrol - INFO - Idle timeout reached after 10 minute(s). Turning heaters off prior to shutting off PSU.
>      2026-10-18 18:12:53,033 - octoprint.plugins.psucontrol - INFO - Turning off heater: o
>      ...
>      ValueError: heater must match "tool", "tool([0-9])", "bed" or "chamber": o
>      ```
>      
>      
>          
>        
>      `Switching PSU Off` was never logged, and the PSU stayed on until it was switched off by hand the next morning.
>      Suggested fix in `_wait_for_heaters()`: only act on valid heater names (`tool\d*`, `bed`, `chamber`), and/or wrap `self._printer.set_temperature()` in a try/except. That way a single bad entry can't stop the PSU from turning off.

Hi @matonb, 

I have a branch in which I've been making changes.  I already added error handling to catch exceptions from set_temperature so the issue is likely already handled.   If you want to give it a try, if it does not solve your problem I am happy to enhance the code to give it a try.

Repo: https://github.com/jeffgregx2/OctoPrint-PSUControl

-Jeff

matonb, 2026-10-23:
@jeffgregx2 
  Yup that solves the 'phantom' temperature sensor issue I was having
Cheers 👍