kantlivelong/OctoPrint-PSUControl #276: Octoprint 1.11.0 broke PSUControl

akamakamak, 2025-04-25:
I upgraded to the Octoprint 1.11.0 and although the plugin can turn the printer on and off, the Octoprint will not be able to connect to the printer.
Had same issues on XL and MK4S

andimae77, 2025-05-05:
SAme problem here.
Configuration: connected with Dirigera PSU plugin. Switching with Native plug control app has no effects on Octoprint, Switching off/on with PSU control breaks the connection from Octoprint to Printer. Restart of Octoprint is necessary then.

oxivanisher, 2025-05-15:
I think the latest 1.11.1 release containing a fix for https://github.com/OctoPrint/OctoPrint/issues/5151 just fixed this problem. At least for me, I am now able to connect to the printer again even if the initial "connect after power up" did not work.

akamakamak, 2025-05-15:
> I think the latest 1.11.1 release containing a fix for [OctoPrint/OctoPrint#5151](https://github.com/OctoPrint/OctoPrint/issues/5151) just fixed this problem. At least for me, I am now able to connect to the printer again even if the initial "connect after power up" did not work.

Could you please check if the Pinter Shutdown option also works?

oxivanisher, 2025-05-15:
You mean just powering off from the web UI?

oxivanisher, 2025-05-15:
> > You mean just powering off from the web UI? I successfully powered the printer on and off several times with a intentionally short "auto connect" value. The manual connect and power off worked every time. For me, it's now working again like before the 1.11 update.
> 
> I meant the Power Off Option: Automatically turn PSU OFF when idle

I will try, but it will take some time. Its currently printing. :)

akamakamak, 2025-05-15:
> I will try, but it will take some time. Its currently printing. :)

Much appreciated!

oxivanisher, 2025-05-16:
> > > I meant the Power Off Option: Automatically turn PSU OFF when idle
> > 
> > I will try, but it will take some time. Its currently printing. :)
> 
> Much appreciated!

Ok, so I did some tests. Automatic power off on idle is NOT working for me.
BUT to be true, I think this stopped working long ago. I didn't even realize that I had this option set on my printer and it definitely did not automatically power off for a very long time. I would guess, that this was introduced when I switched from the [8-bit prusa mk3 with the reprap based firmware](https://github.com/prusa3d/Prusa-Firmware) to the prusa mk4 [32-bit firmware firmware-buddy](https://github.com/prusa3d/Prusa-Firmware-Buddy). There are several issues open on octoprint and in the prusa repositories for serial connection "stuff". In general its working, but several things are not as with the old firmware. This is a ongoing thing where everyone is pointing at the other and I won't go deeper into it. Read [this thread](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3846#issuecomment-2154700645) if you are interested.
But it could just be a result of this incompatibility why this is no longer working but I think this should be its own issue, since the auto connect is absolutely working for me now.

leoheck, 2025-05-23:
I am using the latest 1.11.1 after installing it fresh. PSU control is not working for me too on a Raspbian 12.

I am using GPIO14. Setting it as output during the boot.
And also disabling the serial on it. 

After using the toolbar button to toggle the state of the pin, the GPIO14 reverts back to input. 

Any ideas on how to workaround this?